rock-core/tools-syskit #43: Error during registration of remote process server

maltewi, 2015-03-24:
Configuring a remote process server like this:
```
Syskit.conf.process_server 'vision_cpu', vision_cpu_hostname
```
leads to an error:
```
/home/sempromuser/RockDev/tools/syskit/lib/syskit/roby_app/configuration.rb:587:in `connect_to_orocos_process_server': uninitialized constant Orocos::RemoteProcesses::Client (NameError)
	from /home/sempromuser/RockDev/tools/syskit/lib/syskit/roby_app/configuration.rb:530:in `process_server'
	from /home/sempromuser/RockDev/bundles/besman_aila/config/init.rb:50:in `<top (required)>'
	from /home/sempromuser/RockDev/tools/roby/lib/roby/app.rb:860:in `require'
	from /home/sempromuser/RockDev/tools/roby/lib/roby/app.rb:860:in `block in require'
	from /home/sempromuser/RockDev/tools/roby/lib/roby/app.rb:840:in `isolate_load_errors'
	from /home/sempromuser/RockDev/tools/roby/lib/roby/app.rb:858:in `require'
	from /home/sempromuser/RockDev/tools/roby/lib/roby/app.rb:956:in `load_base_config'
	from /home/sempromuser/RockDev/tools/roby/lib/roby/app.rb:976:in `base_setup'
	from /home/sempromuser/RockDev/tools/roby/lib/roby/standalone.rb:5:in `<top (required)>'
	from /usr/lib/ruby/1.9.1/rubygems/custom_require.rb:36:in `require'
	from /usr/lib/ruby/1.9.1/rubygems/custom_require.rb:36:in `require'
	from /home/sempromuser/RockDev/tools/syskit/lib/syskit/scripts/instanciate.rb:1:in `<top (required)>'
	from /usr/lib/ruby/1.9.1/rubygems/custom_require.rb:36:in `require'
	from /usr/lib/ruby/1.9.1/rubygems/custom_require.rb:36:in `require'
	from /home/sempromuser/RockDev/tools/syskit/bin/syskit:19:in `<main>'
```

maltewi, 2015-03-24:
I added some 'requires' to orocos.rb that includes name module. But still there is a problem. In ```orocos.rb/remote_processes/client.rb:112``` the function call ```Marshal.load(socket)``` fails with the error message:

```
/home/sempromuser/RockDev/tools/orocos.rb/lib/orocos/remote_processes/client.rb:113:in `load': end of file reached (EOFError)
	from /home/sempromuser/RockDev/tools/orocos.rb/lib/orocos/remote_processes/client.rb:113:in `info'
	from /home/sempromuser/RockDev/tools/orocos.rb/lib/orocos/remote_processes/loader.rb:16:in `initialize'
	from /home/sempromuser/RockDev/tools/orocos.rb/lib/orocos/remote_processes/client.rb:88:in `new'
	from /home/sempromuser/RockDev/tools/orocos.rb/lib/orocos/remote_processes/client.rb:88:in `initialize'
	from /home/sempromuser/RockDev/tools/syskit/lib/syskit/roby_app/configuration.rb:588:in `new'
	from /home/sempromuser/RockDev/tools/syskit/lib/syskit/roby_app/configuration.rb:588:in `connect_to_orocos_process_server'
	from /home/sempromuser/RockDev/tools/syskit/lib/syskit/roby_app/configuration.rb:531:in `process_server'
	from /home/sempromuser/RockDev/bundles/besman_aila/config/init.rb:50:in `<top (required)>'
	from /home/sempromuser/RockDev/tools/roby/lib/roby/app.rb:860:in `require'
	from /home/sempromuser/RockDev/tools/roby/lib/roby/app.rb:860:in `block in require'
	from /home/sempromuser/RockDev/tools/roby/lib/roby/app.rb:840:in `isolate_load_errors'
	from /home/sempromuser/RockDev/tools/roby/lib/roby/app.rb:858:in `require'
	from /home/sempromuser/RockDev/tools/roby/lib/roby/app.rb:956:in `load_base_config'
	from /home/sempromuser/RockDev/tools/roby/lib/roby/app.rb:976:in `base_setup'
	from /home/sempromuser/RockDev/tools/roby/lib/roby/standalone.rb:5:in `<top (required)>'
	from /usr/lib/ruby/1.9.1/rubygems/custom_require.rb:36:in `require'
	from /usr/lib/ruby/1.9.1/rubygems/custom_require.rb:36:in `require'
	from /home/sempromuser/RockDev/tools/syskit/lib/syskit/scripts/instanciate.rb:1:in `<top (required)>'
	from /usr/lib/ruby/1.9.1/rubygems/custom_require.rb:36:in `require'
	from /usr/lib/ruby/1.9.1/rubygems/custom_require.rb:36:in `require'
	from /home/sempromuser/RockDev/tools/syskit/bin/syskit:19:in `<main>'
```

orocos.rb is on master.

doudou, 2015-03-24:
That would smell like a dying process server. Nothing on the process server output ?

maltewi, 2015-03-24:
Right.. it's dying. The installation on the remote PC is pretty messed up...

Should make pull requests on orocos.rb and syskit to where I included the missing ```require 'some_file'``` calls?

doudou, 2015-03-24:
> Should make pull requests on orocos.rb and syskit to where I included the missing require 'some_file' calls?

That would be great

doudou, 2015-08-31:
Seems to be fixed, I close.

closed 2015-08-31 by doudou