rock-core/tools-syskit #51: Compositions getting restarted unexpectedly

maltewi, 2015-04-27:
I'm observing a behaviour in syskit, where the start event of compositions get executed when they should not.

Imagine a Composition A that’s running. Now via ActionScript  another Compositions T, that has nothing to do with A gets started. What happens is, that A gets stopped and started again when T is being called.

Now follows small test program illustrating the issue more detailed.

We have a composition ```CmpA``` that looks like this:
```
    class CmpA < Syskit::Composition
        on :start do |event|
            Robot.info("#{self.name}: Starting")
        end

        poll do
            @polled ||= false
            if not @polled
                Robot.info("#{self.name}: Pollin' around")
                @polled = true
            end
        end

        on :stop do |event|
            Robot.info("#{self.name}: Stopping")
        end
    end

    profile "P" do
        define 'test', Compositions::CmpA
    end
```
It reacts to an event with printing what has just happened together with information about ```self```. 

There is also another Compositions which is used in an ActionScript:
```
class T < Syskit::Composition 
	terminates

	event :start do |event|
		puts "#{self.name}: Starting"
		emit :start
	end

	script do
		poll do
			success_event.emit
		end
	end

	on :stop do |event|
		puts  "#{self.name}: Stopping"
	end

end

class Main < Roby::Actions::Interface
	use_profile Profiles::P

    describe("A Dummy Action")
    action_script "dummy" do
    	t = task T
    	execute t
    	emit success_event
    end
end
```

I start my program with ```syskit run test_def!```, then wait some time, and via Syskit Shell send the command ```dummy!```. Here is the console output:
```
16:08:01.746 (Robot) loaded Roby on ruby 1.9.3p484 (2013-11-22 revision 43786) [x86_64-linux]
16:08:01.746 (Robot) ready
16:08:01.751 (Robot) done initialization
16:08:01.794 (Robot) Compositions::CmpA:0x36e2190: Starting
16:08:01.795 (Robot) Compositions::CmpA:0x36e2190: Pollin' around
16:08:01.950 (Roby) new interface client: localhost:42666
#NOW I SEND DUMMY
T:0x2381330: Starting
T:0x2381330: Stopping
16:09:17.011 (Robot) Compositions::CmpA:0x36dc5d8: Starting
16:09:17.011 (Robot) Compositions::CmpA:0x36dc5d8: Pollin' around
16:09:17.017 (Robot) Compositions::CmpA:0x36e2190: Stopping
16:09:17.123 (Roby) new interface client: localhost:42751
```

I would have expected to not see what comes after ```T:0x2381330: Stopping```.

Any ideas?

maltewi, 2015-04-27:
I made another test, I guess the result is relevant. I modified ```T``` to be a ```Roby::Task``` (Not ```Syskit::Composition```). In this case, the behavior is as expected.

Here is the modified snipped:
```
class T < Roby::Task 
	terminates

	event :start do |event|
		puts "#{self.name}: Starting"
		emit :start
	end

	script do
		poll do
			success_event.emit
		end
	end

	on :stop do |event|
		puts  "#{self.name}: Stopping"
	end

end

class Main < Roby::Actions::Interface
	use_profile Profiles::P

	describe("t").
	returns T
	def make_t()
		T.new
	end

    describe("A Dummy Action")
    action_script "dummy" do
    	t = task make_t
    	execute t
    	emit success_event
    end
end
```

doudou, 2015-04-28:
This looks like a bug in the merger. I guess that if you looked at the roby logs, you would see that your composition got killed and a new one started (which it should not).

doudou, 2017-07-18:
There was indeed a bug there. It got fixed a while back.

closed 2017-07-18 by doudou