rock-core/tools-syskit #62: syskit browse doesn't work

gustavoneves12, 2015-08-13:
When I execute the command syskit browse I got the error below:

```
Bundles[INFO]: Active bundles: tutorial
15:07:37.522 (Robot) loaded Roby application in 0.016 seconds

= /home/gustavoneves/rock-robotics/tools/syskit/lib/syskit/gui/model_browser.rb:27:in `method_missing': undefined local variable or method `btn_reload_models' for #<Syskit::GUI::ModelBrowser:0x00000003258d90> (NameError)
= Backtrace
| 
|   /home/gustavoneves/rock-robotics/tools/syskit/lib/syskit/gui/model_browser.rb:27:in `method_missing'
|   /home/gustavoneves/rock-robotics/tools/syskit/lib/syskit/gui/model_browser.rb:27:in `initialize'
|   /home/gustavoneves/rock-robotics/tools/syskit/lib/syskit/scripts/browse.rb:40:in `new'
|   /home/gustavoneves/rock-robotics/tools/syskit/lib/syskit/scripts/browse.rb:40:in `block in <top (required)>'
|   /home/gustavoneves/rock-robotics/tools/syskit/lib/syskit/scripts/common.rb:267:in `block in run'
|   /home/gustavoneves/rock-robotics/tools/roby/lib/roby/exceptions.rb:420:in `display_exception'
|   /home/gustavoneves/rock-robotics/tools/syskit/lib/syskit/scripts/common.rb:264:in `run'
|   /home/gustavoneves/rock-robotics/tools/syskit/lib/syskit/scripts/browse.rb:37:in `<top (required)>'
|   /usr/share/rvm/rubies/ruby-2.0.0-p643/lib/ruby/site_ruby/2.0.0/rubygems/core_ext/kernel_require.rb:54:in `require'
|   /usr/share/rvm/rubies/ruby-2.0.0-p643/lib/ruby/site_ruby/2.0.0/rubygems/core_ext/kernel_require.rb:54:in `require'
|   /home/gustavoneves/rock-robotics/tools/syskit/bin/syskit:19:in `<main>'
= 
/home/gustavoneves/rock-robotics/install/lib/ruby/2.0.0/typelib/registry_export.rb:10:in `reset_registry_export': wrong number of arguments (0 for 2..3) (ArgumentError)
	from /home/gustavoneves/rock-robotics/tools/orocos.rb/lib/orocos/default_loader.rb:41:in `clear'
	from /home/gustavoneves/rock-robotics/tools/syskit/lib/syskit/roby_app/plugin.rb:486:in `clear_models'
	from /home/gustavoneves/rock-robotics/tools/roby/lib/roby/app.rb:658:in `block in call_plugins'
	from /home/gustavoneves/rock-robotics/tools/roby/lib/roby/app.rb:650:in `block in each_responding_plugin'
	from /home/gustavoneves/rock-robotics/tools/roby/lib/roby/app.rb:643:in `block in each_plugin'
	from /home/gustavoneves/rock-robotics/tools/roby/lib/roby/app.rb:642:in `each'
	from /home/gustavoneves/rock-robotics/tools/roby/lib/roby/app.rb:642:in `each_plugin'
	from /home/gustavoneves/rock-robotics/tools/roby/lib/roby/app.rb:649:in `each_responding_plugin'
	from /home/gustavoneves/rock-robotics/tools/roby/lib/roby/app.rb:657:in `call_plugins'
	from /home/gustavoneves/rock-robotics/tools/roby/lib/roby/app.rb:1925:in `clear_models'
	from /home/gustavoneves/rock-robotics/tools/roby/lib/roby/app.rb:1405:in `cleanup'
	from /home/gustavoneves/rock-robotics/tools/syskit/lib/syskit/scripts/common.rb:273:in `run'
	from /home/gustavoneves/rock-robotics/tools/syskit/lib/syskit/scripts/browse.rb:37:in `<top (required)>'
	from /usr/share/rvm/rubies/ruby-2.0.0-p643/lib/ruby/site_ruby/2.0.0/rubygems/core_ext/kernel_require.rb:54:in `require'
	from /usr/share/rvm/rubies/ruby-2.0.0-p643/lib/ruby/site_ruby/2.0.0/rubygems/core_ext/kernel_require.rb:54:in `require'
	from /home/gustavoneves/rock-robotics/tools/syskit/bin/syskit:19:in `<main>'
```

gustavoneves12, 2015-08-14:
I did a local merge between the master and syskit_ide and this error was solved.

doudou, 2015-08-14:
I keep it open until syskit_ide is merged in master

closed 2015-08-31 by doudou