rock-core/tools-syskit #65: ide: don't show some types

gustavoneves12, 2015-08-14:
I couldn't to find Base::PoseSrv when I execute syskit ide command.

doudou, 2015-08-17:
Base::PoseSrv is a deprecated naming (I guess you got the name from the tutorials, which are unfortunately outdated). It now aliases to Rock::Services::Pose

In other words, there's no service called "Base::PoseSrv", but you can use Base::PoseSrv in the syskit models, it should "simply" be an alias to Rock::Services::Pose. This aliasing cannot be shown by syskit ide, really (well, it could, but given that it's useful mainly in transitions I personally see it mostly as a enhancement and not a bug).

doudou, 2015-08-17:
Opened #66 to keep track of the functionality, closing this now.

closed 2015-08-17 by doudou