rock-core/tools-syskit #93: Changing the configuration of a CompositionChild

D-Alex, 2015-12-02:
The following code as no effect on the configuration:

    define 'control',Compositions::Control::BaseControl.use('controlled_system' => viator_model_dev)
    control_def.controller_child.with_conf('default','gazebo')

The following code works but feels wrong:

    define 'control',Compositions::Control::BaseControl.use('controlled_system' => viator_model_dev, 'controller' => OroGen::Skid4Control::SimpleController.with_conf('default','gazebo'))

doudou, 2015-12-07:
> The following code works but feels wrong:

Does not feel wrong to me ... and is actually the right way. Definitions are only based on dependency injection, there is no in-place modification, because these changes cannot be meaningfully propagated to definitions that would be using it.

closed 2015-12-07 by doudou

D-Alex, 2015-12-07:
That means with_conf should ideally be a private/protected member method?

doudou, 2015-12-07:
> That means with_conf should ideally be a private/protected member method?

Not really. The only thing that could be maybe an improvement is to make _child non-accessible on Definition objects. But, then, it makes the whole InstanceRequirements API non-uniform which sucks when reasoning about requirements.

Honestly, I personally see it more as a documentation issue than a code issue.