rock-core/tools-syskit #96: Generating a Composition Error

rafaelsaback, 2016-02-19:
I was generating a new composition, but for some reason syskit was disconsidering a " _ " character of the name I used (check the image below). This has happened with different names, but basically only when I had " _ " + character + " _ " OR had " _ " + character + character + " _ ".

So I had to correct the files by hand later.

[image]

doudou, 2016-02-23:
Workaround: pass the target class name instead of the file name (ModelPredictiveControl/ConstantAlignedVelZControl) which should have worked. (Disclaimer: not tested).

rafaelsaback, 2016-02-24:
Using the following command:

```
syskit gen cmp ModelPredictiveControl/ConstantAlignedVelZControl
```

Generates, for example, the following folder:

```
models/compositions/model_predictive_control/constantalignedvelzcontrol.rb
```

Am I missing something?

doudou, 2018-05-23:
`syskit gen` in master stopped trying to guess between models and files. The argument is now always a file name from which the class name is inferred.

closed 2018-05-23 by doudou