rock-core/tools-syskit #101: Syskit Shell not working with websocket 1.2.3

2maz, 2016-07-15:
Syskit shell is not opening when using websocket 1.2.3 - using 1.2.2 is doing fine though.

doudou, 2016-07-18:
I'm running just fine on 1.2.3. Any more info ? (backtrace, ...)

doudou, 2016-07-18:
Ruby version would be useful too.

2maz, 2016-07-19:
Ubuntu 14.04/ruby2.0 -- syskit shell sits in the while loop of  Roby::Interface::DRobyChannel.read_packet

What is to be observed when syskit shell is opened:
syskit run --> with websocket 1.2.2: calls Roby::Interface::DRobyChannel.read_packet and then continues with calls to Roby::Interface::DRobyChannel.write_packet -- with websocket 1.2.3 this does not happen

doudou, 2017-07-18:
Was a 2.0-specific error, I don't plan to investigate it further. 1.2.4 works like a charm on Ruby 2.1+

closed 2017-07-18 by doudou