rock-core/tools-syskit #122: Manage the LoggingConfiguration of a syskit instance

g-arjones, 2017-10-26:
Is there an API to manage the `LoggingConfiguration` from an external process? I had a look at `Roby::Interface::Async::Interface` but did not find anything I could use there. I wanted to develop a GUI that allowed the user to manage that during the operation. Any ideas?

doudou, 2017-10-26:
Currently, you can only enable/disable log groups from the sync interface. Check the enable_log_group and disable_log_group methods in `lib/syskit/shell.rb`. This is available as the `syskit` subcommand of the main interface (in the shell, `syskit.enable_log_group 'group_name'`)

I think a UI sounds nice - if you could implement it as part of the IDE, that would be even better ;-). The easy-to-get, in-my-opinion useful features would be:
 - list ports that are *not* logged (since all ports are logged by default)
 - check what enabling/disabling a group would enable/disable
 - enable or disable log groups

g-arjones, 2017-10-26:
> Check the enable_log_group and disable_log_group methods in lib/syskit/shell.rb

Did you mean `shell_interface.rb`? The only thing in enable/disable_log_group is a call to a singleton method. Could you please provider further info on how do I get an instance of the sync interface (syskit is still a blackbox to me) ?

> if you could implement it as part of the IDE, that would be even better ;-)

Of course, my idea is to add that as a widget in vizkit and then add it to syskit ide (like the task inspector thing).

> The easy-to-get, in-my-opinion useful features would be:

I had something like that in mind which seemed pretty trivial when I saw `Syskit.conf.logs` but then could not figure it out how to get there.

doudou, 2017-10-26:
`Roby::Interface::Async::Interface#client` is an `Roby::Interface::Client`. The former should really be called Client instead of Interface, actually. Calling a subcommand such as `syskit` on that will return an object on which you can call the commands (SubcommandClient), e.g.

~~~
client.syskit.enable_log_group('args')
~~~

will call the corresponding method on the remote interface. However, the call is synchronous, which means tha it will (1) block the IDE until it returns and (2) will raise exceptions (which could kill the IDE). There is already a bunch of things that are done synchronously (such as e.g. #quit), so it would not change the situation much. Nonetheless making all of this async is on the list of "nice to do".

> Of course, my idea is to add that as a widget in vizkit and then add it to syskit ide (like the task inspector thing).

I would implement such very-Syskit-specific things in Syskit, really. Adding a vizkit-to-syskit dependency would be a headache.

g-arjones, 2017-10-26:
So, if I got that right, unless it is part of `Syskit::ShellInterface` (lib/syskit/shell_interface.rb) it is not available remotely (or am I missing something?). I can see methods to enable/disable log groups but I can't see anything that would give me the `groups` hash from `Syskit.conf.logs.groups` (so I could see which ones are enabled) nor anything that would give me what is part of a given group (to show to the user what enabling/disabling that group really means). All that mean I have to implement the interface in syskit itself first, right?

> I would implement such very-Syskit-specific things in Syskit, really. Adding a vizkit-to-syskit dependency would be a headache.

Is it ok if I then add that as a widget in `Syskit::GUI` (lib/syskit/gui) so we can reuse it in our operation gui?

doudou, 2017-10-26:
You got it right, you have to first create an API on the shell interface. Most standard Ruby types plus some of the Syskit types (such as task objects) can be transferred.

> Is it ok if I then add that as a widget in Syskit::GUI (lib/syskit/gui) so we can reuse it in our operation gui?

That would be the best course of action. Widget in Syskit::GUI that is given an `Async::Interface` at construction time. Use in both the operation GUI and IDE.

g-arjones, 2017-10-26:
Perfect!

How do you feel about a `pull_logging_config()` and `push_logging_config(logs)` instead of having getters/setters for each thing we want to be able to manage? So, instead of `enable_log_group()` and `get_log_group_state()` the user would pull_logging_config(), set everything he wishes in the local copy of the model and then push it back...

doudou, 2017-10-27:
Generally speaking, I don't see a problem with what you propose, especially given that calling many remote methods is always horribly slow ... 

But it IMO depends on what you include in "config". I.e. I would still attempt to keep a sane API.

(Style note: I never name methods `get_something`, stick to that within Syskit please)

doudou, 2017-10-27:
You get this as soon as you connect ?

g-arjones, 2017-10-27:
Nope.. As soon as try to access `syskit` object from `Roby::Interface::Client`

Steps to reproduce:

1. Run `syskit run -rgazebo -c` (wait for it to finish loading)
2. Run `syskit shell`
3. In syskit shell, run `client.syskit` (freezes and errors are shown in syskit's terminal)

doudou, 2017-10-27:
> Is this a bug?

Yes. I forgot, but remote access to subcommands is part of the post-FF roby3/syskit2 branches. I've tried to push Gustavo to test and migrate to the current syskit2/roby3 branches, but it seems that I failed.

There's quite a bit of goodies there ...

In the meantime, you could call the plumbing. `call` is the thing that calls the remote behind the scenes, with the first argument being the subcommand path (i.e. `['syskit']` in our case).

gustavoneves12, 2017-10-27:
@doudou, I did not get time to test the stuff you asked me. As soon as possible, I want to test and migrate to the current syskit2/roby3 branches.

g-arjones, 2017-11-07:
So, just started to work on this... `Syskit::RobyApp::LoggingConfiguration` can't be marshalled (because of `@configuration_log`) so I am going to create a marshallable model containing only stuff that we want to be able to manage remotely. First, I thought of putting it in `Syskit::RemoteModels::LoggingConfiguration` but then I saw that there is a `Syskit::Models`, now not so sure anymore... Thoughts?

doudou, 2017-11-08:
I really would start small, and only allow to enable/disable log groups. In which case, the only thing you need for the interface is a hash from a group name to whether a group is enabled/disabled, possibly with a documentation string (there currently is none in the definition AFAIK, so it would need to be added there)

That's the core functionality. Once you've built a UI for that, what else do you want to add ?

g-arjones, 2017-11-08:
> I really would start small, and only allow to enable/disable log groups.

That's what I am doing...

> the only thing you need for the interface is a hash from a group name to whether a group is enabled/disabled

That would be enough but don't you think it would make the code cleaner and easier to manage if we defined a class with proper accessors to manage it? Specially if we move on and decide to do more then just enable/disable log groups.

> what else do you want to add ?

I liked your suggestions: 

- list ports that are not logged (since all ports are logged by default)
- check what enabling/disabling a group would enable/disable

doudou, 2017-11-08:
> don't you think it would make the code cleaner and easier to manage if we defined a class with proper accessors to manage it? 

Fair point ... Even after may years of Ruby, I still have the bad reflex of relying on Hashes.

There are two ways to handle this:
 - you define yourself a Struct specifically for the purpose of the Interface (under ShellInterface) and do the "marshalling" there.
 - you get LoggerConfiguration droby-marshallable.

I would go for (1), since being able to marshall the whole configuration struct is (a) not needed right now and (b) wouldn't be enough if you want to implement what I suggested.

> > what else do you want to add ?
> I liked your suggestions:
> 
> - list ports that are not logged (since all ports are logged by default)
> - check what enabling/disabling a group would enable/disable

Both things are better computed on the Syskit side anyways, which you would then embed in said struct.

g-arjones, 2017-11-22:
@doudou Do you want to keep this one open?

doudou, 2017-11-22:
Nope ...

closed 2017-11-22 by doudou