rock-core/tools-syskit #144: Periodicity of composition's poll

joaobrittoneto, 2017-12-12:
I use a composition to generate input commands for the [auv_control](https://github.com/rock-control/control-orogen-auv_control). It happens that the auv_control was expecting to receive `cmd_in` (provide by `Dive::Compositions::SinusoidSignalGenerators::WorldPos`) with at least 1s. 

It happens that sometimes, the poll block of the compositions are called with a ~2s interval (16:28:51, 16:28:53, 16:28:55).
Just increasing the timeout would fix the problem, but I'd like to understand what Syskit is doing in the meantime. 
Is there a way to guarantee the periodicity of a composition's poll?

Here is the data provided by `roby-log decode live-events.log`
```
2017-12-05 16:28:51 -0300 timepoint_group_start  #<DRobyID:2>  MAIN  poll block of FlatFish::Compositions::ModelEstimation::OneDofModelEstimationWithTimeout:0x7f5aed91e318<id:6152>(estimator_timeout: 300, mission_name: linear_y_model_estimation, timeout_action: FlatFish::Actions::Live::Navigation.mission_timeout(mission_name => linear_y_model_estimation, mission_timeout => 300))
2017-12-05 16:28:51 -0300 timepoint_group_end  #<DRobyID:2>  MAIN  poll block of FlatFish::Compositions::ModelEstimation::OneDofModelEstimationWithTimeout:0x7f5aed91e318<id:6152>(estimator_timeout: 300, mission_name: linear_y_model_estimation, timeout_action: FlatFish::Actions::Live::Navigation.mission_timeout(mission_name => linear_y_model_estimation, mission_timeout => 300))
2017-12-05 16:28:51 -0300 timepoint_group_start  #<DRobyID:2>  MAIN  poll block of FlatFish::Tasks::MissionTimeout:0x82b5a40<id:6213>(mission_name: linear_y_model_estimation, mission_timeout: 300)
2017-12-05 16:28:51 -0300 timepoint_group_end  #<DRobyID:2>  MAIN  poll block of FlatFish::Tasks::MissionTimeout:0x82b5a40<id:6213>(mission_name: linear_y_model_estimation, mission_timeout: 300)
2017-12-05 16:28:51 -0300 timepoint_group_start  #<DRobyID:2>  MAIN  poll block of FlatFish::Actions::Navigation::SinusoidalMovement:0x7f5aef553420<id:6243>()
2017-12-05 16:28:51 -0300 timepoint_group_end  #<DRobyID:2>  MAIN  poll block of FlatFish::Actions::Navigation::SinusoidalMovement:0x7f5aef553420<id:6243>()
2017-12-05 16:28:51 -0300 timepoint_group_start  #<DRobyID:2>  MAIN  poll block for Dive::Compositions::SinusoidSignalGenerators::WorldPosX:0x7f5aec0e0ff8<id:6367>(amplitude: 0, conf: ["default"], frequency: 0, offset: 0.5613446366334031, orocos_name: sinusoid_signal_world_pos_x)
2017-12-05 16:28:51 -0300 timepoint_group_end  #<DRobyID:2>  MAIN  poll block for Dive::Compositions::SinusoidSignalGenerators::WorldPosX:0x7f5aec0e0ff8<id:6367>(amplitude: 0, conf: ["default"], frequency: 0, offset: 0.5613446366334031, orocos_name: sinusoid_signal_world_pos_x)
2017-12-05 16:28:51 -0300 timepoint_group_start  #<DRobyID:2>  MAIN  poll block for Dive::Compositions::SinusoidSignalGenerators::WorldPosY:0x7f5aee693b38<id:6368>(amplitude: 2, conf: ["default"], frequency: 0.2, offset: -2.2394328715524163, orocos_name: sinusoid_signal_world_pos_y)
2017-12-05 16:28:51 -0300 timepoint_group_end  #<DRobyID:2>  MAIN  poll block for Dive::Compositions::SinusoidSignalGenerators::WorldPosY:0x7f5aee693b38<id:6368>(amplitude: 2, conf: ["default"], frequency: 0.2, offset: -2.2394328715524163, orocos_name: sinusoid_signal_world_pos_y)
2017-12-05 16:28:51 -0300 timepoint_group_start  #<DRobyID:2>  MAIN  poll block for Dive::Compositions::SinusoidSignalGenerators::WorldPosZ:0x7f5aef3c55b8<id:6369>(amplitude: 0, conf: ["default"], frequency: 0, offset: -3.066761269564217, orocos_name: sinusoid_signal_world_pos_z)
2017-12-05 16:28:51 -0300 timepoint_group_end  #<DRobyID:2>  MAIN  poll block for Dive::Compositions::SinusoidSignalGenerators::WorldPosZ:0x7f5aef3c55b8<id:6369>(amplitude: 0, conf: ["default"], frequency: 0, offset: -3.066761269564217, orocos_name: sinusoid_signal_world_pos_z)
2017-12-05 16:28:51 -0300 timepoint_group_start  #<DRobyID:2>  MAIN  poll block for Dive::Compositions::SinusoidSignalGenerators::WorldPosYaw:0x7f5aeeb0aa40<id:6373>(amplitude: 0, conf: ["default"], frequency: 0, offset: 0.006001968664297033, orocos_name: sinusoid_signal_world_pos_yaw)
2017-12-05 16:28:51 -0300 timepoint_group_end  #<DRobyID:2>  MAIN  poll block for Dive::Compositions::SinusoidSignalGenerators::WorldPosYaw:0x7f5aeeb0aa40<id:6373>(amplitude: 0, conf: ["default"], frequency: 0, offset: 0.006001968664297033, orocos_name: sinusoid_signal_world_pos_yaw)
2017-12-05 16:28:51 -0300 timepoint  #<DRobyID:2>  MAIN  external_events
2017-12-05 16:28:51 -0300 scheduler_report_pending_non_executable_task  partially instanciated  #<RemoteDRobyID @#<DRobyID:6637>>
2017-12-05 16:28:51 -0300 scheduler_report_pending_non_executable_task  partially instanciated  #<RemoteDRobyID @#<DRobyID:6673>>
2017-12-05 16:28:51 -0300 scheduler_report_pending_non_executable_task  partially instanciated  #<RemoteDRobyID @#<DRobyID:6709>>
2017-12-05 16:28:51 -0300 scheduler_report_pending_non_executable_task  partially instanciated  #<RemoteDRobyID @#<DRobyID:6745>>
2017-12-05 16:28:51 -0300 scheduler_report_trigger  #<RemoteDRobyID @#<DRobyID:6640>>
2017-12-05 16:28:51 -0300 scheduler_report_trigger  #<RemoteDRobyID @#<DRobyID:6676>>
2017-12-05 16:28:51 -0300 scheduler_report_trigger  #<RemoteDRobyID @#<DRobyID:6712>>
2017-12-05 16:28:51 -0300 scheduler_report_trigger  #<RemoteDRobyID @#<DRobyID:6748>>
2017-12-05 16:28:51 -0300 timepoint  #<DRobyID:2>  MAIN  scheduler
2017-12-05 16:28:51 -0300 timepoint_group_start  #<DRobyID:2>  MAIN  poll_state_events
2017-12-05 16:28:51 -0300 timepoint_group_end  #<DRobyID:2>  MAIN  poll_state_events
2017-12-05 16:28:51 -0300 timepoint  #<DRobyID:2>  MAIN  propagation_handlers
2017-12-05 16:28:51 -0300 timepoint_group_start  #<DRobyID:2>  MAIN  event_propagation_phase
2017-12-05 16:28:51 -0300 generator_propagate_events  false  #<Set:0x0000000390b340>  #<RemoteDRobyID @#<DRobyID:6640>>
2017-12-05 16:28:51 -0300 generator_propagate_events  true  #<Set:0x0000000390b250>  #<RemoteDRobyID @#<DRobyID:6640>>
2017-12-05 16:28:51 -0300 generator_fired  #<Roby::DRoby::V5::EventDumper::DRoby:0x0000000390b110>
2017-12-05 16:28:51 -0300 generator_propagate_events  false  #<Set:0x0000000390acb0>  #<RemoteDRobyID @#<DRobyID:6676>>
2017-12-05 16:28:51 -0300 generator_propagate_events  true  #<Set:0x0000000390abc0>  #<RemoteDRobyID @#<DRobyID:6676>>
2017-12-05 16:28:51 -0300 generator_fired  #<Roby::DRoby::V5::EventDumper::DRoby:0x0000000390aaa8>
2017-12-05 16:28:51 -0300 generator_propagate_events  false  #<Set:0x0000000390a800>  #<RemoteDRobyID @#<DRobyID:6712>>
2017-12-05 16:28:51 -0300 generator_propagate_events  true  #<Set:0x0000000390a710>  #<RemoteDRobyID @#<DRobyID:6712>>
2017-12-05 16:28:51 -0300 generator_fired  #<Roby::DRoby::V5::EventDumper::DRoby:0x0000000390a5f8>
2017-12-05 16:28:51 -0300 generator_propagate_events  false  #<Set:0x0000000390a350>  #<RemoteDRobyID @#<DRobyID:6748>>
2017-12-05 16:28:51 -0300 generator_propagate_events  true  #<Set:0x0000000390a260>  #<RemoteDRobyID @#<DRobyID:6748>>
2017-12-05 16:28:51 -0300 generator_fired  #<Roby::DRoby::V5::EventDumper::DRoby:0x0000000390a148>
2017-12-05 16:28:51 -0300 scheduler_report_pending_non_executable_task  partially instanciated  #<RemoteDRobyID @#<DRobyID:6637>>
2017-12-05 16:28:51 -0300 scheduler_report_pending_non_executable_task  partially instanciated  #<RemoteDRobyID @#<DRobyID:6673>>
2017-12-05 16:28:51 -0300 scheduler_report_pending_non_executable_task  partially instanciated  #<RemoteDRobyID @#<DRobyID:6709>>
2017-12-05 16:28:51 -0300 scheduler_report_pending_non_executable_task  partially instanciated  #<RemoteDRobyID @#<DRobyID:6745>>
2017-12-05 16:28:51 -0300 timepoint  #<DRobyID:2>  MAIN  scheduler
2017-12-05 16:28:51 -0300 timepoint_group_start  #<DRobyID:2>  MAIN  poll_state_events
2017-12-05 16:28:51 -0300 timepoint_group_end  #<DRobyID:2>  MAIN  poll_state_events
2017-12-05 16:28:51 -0300 timepoint_group_start  #<DRobyID:2>  MAIN  syskit:connection_management_update
2017-12-05 16:28:51 -0300 timepoint_group_end  #<DRobyID:2>  MAIN  syskit:connection_management_update
2017-12-05 16:28:51 -0300 timepoint_group_start  #<DRobyID:2>  MAIN  syskit:apply_requirement_modifications
2017-12-05 16:28:51 -0300 timepoint_group_end  #<DRobyID:2>  MAIN  syskit:apply_requirement_modifications
2017-12-05 16:28:51 -0300 timepoint_group_end  #<DRobyID:2>  MAIN  event_propagation_phase
2017-12-05 16:28:51 -0300 timepoint_group_start  #<DRobyID:2184>  syskit-async-resolution  syskit-async-resolution
[... cut ...]
2017-12-05 16:28:55 -0300 timepoint_group_start  #<DRobyID:2>  MAIN  poll block for Dive::Compositions::SinusoidSignalGenerators::WorldPosYaw:0x7f5aeeb0aa40<id:6373>(amplitude: 0, conf: ["default"], frequency: 0, offset: 0.006001968664297033, orocos_name: sinusoid_signal_world_pos_yaw)
2017-12-05 16:28:55 -0300 timepoint_group_end  #<DRobyID:2>  MAIN  poll block for Dive::Compositions::SinusoidSignalGenerators::WorldPosYaw:0x7f5aeeb0aa40<id:6373>(amplitude: 0, conf: ["default"], frequency: 0, offset: 0.006001968664297033, orocos_name: sinusoid_signal_world_pos_yaw)
2017-12-05 16:28:55 -0300 timepoint  #<DRobyID:2>  MAIN  external_events
```

doudou, 2017-12-13:
> I use a composition to generate input commands for the auv_control

`poll` is called at the periodicity of the [Syskit event loop](https://doudou.github.io/rock_website/runtime_overview/event_loop.html). Its periodicity is not guaranteed.

**However**, the plan generation is meant to be asynchronous, leaving the event loop free to react to things (and call `poll`), The log you've just shown tends to prove otherwise - the 2s delay you're seeing seem to be related to the network generation since there are no timepoints related to the event loop within the 2s period. This is seems to be a Syskit performance / setup bug that needs to be investigated.

Moreover, `ConstantGenerator` does the generation within a thread to guarantee a better periodicity. While this may be affected by the same issue, I would suggest to refactor ConstantGenerator into a ThreadedPortWriter abstract base class where most of the thread-handling would be done, and use this as a basis for your sinusoidal generator. This would definitely improve reactivity. However, it could also be affected by the same thread scheduling bug.

joaobrittoneto, 2017-12-13:
> This is seems to be a Syskit performance / setup bug that needs to be investigated.

At this moment, syskit were starting the cameras.
It seems that the camera driver has some sleeps in the startHook([1](https://github.com/rock-drivers/drivers-orogen-camera_base/blob/master/tasks/Task.cpp#L101),[2](https://github.com/rock-drivers/drivers-camera_interface/blob/master/src/CamInterface.cpp#L179)) and [stopHook](https://github.com/rock-drivers/drivers-orogen-camera_base/blob/master/tasks/Task.cpp#L232).
Would it affect the syskit operation? Would Syskit wait for the start / stopHook to finalize? 
[image]

doudou, 2017-12-13:
> Would Syskit wait for the start / stopHook to finalize?

The analysis I gave you comes from the log you sent. The network generation seem to be starving the main thread, which means that Syskit almost does not react to incoming events. This is purely internal to Syskit (probably a thread priority thing).

Syskit does not wait for configure/start/stop/cleanup (it waits for them to complete asynchronously).

joaobrittoneto, 2017-12-14:
> The network generation seem to be starving the main thread, which means that Syskit almost does not react to incoming events

It seems to be the case. There are other cases where the poll block take some time to be called during a network generation.

> Moreover, ConstantGenerator does the generation within a thread to guarantee a better periodicity. While this may be affected by the same issue, I would suggest to refactor ConstantGenerator into a [...]

It could help. But I wonder if creating a component to provide a dynamic signal would not be a simpler and more robust solution (I imagine this kind of discussion had happened before )

doudou, 2017-12-14:
> It could help. But I wonder if creating a component to provide a dynamic signal would not be a simpler and more robust solution (I imagine this kind of discussion had happened before )

Generally speaking, creating a workflow that does not require a full blown C++ component for generating simple values definitely has some value (on both development *and* deployment time). But YMMV.

At this stage, Syskit is the limiting factor. If you don't have the resources/time to improve the situation, an orogen component will definitely work for your use case.

doudou, 2017-12-15:
For the record, the Syskit bug must be fixed in any case since it affects its reactivity.

joaobrittoneto, 2018-01-02:
> For the record, the Syskit bug must be fixed in any case since it affects its reactivity.

Agreed. I see this bug as a major bottle neck for Syskit. 

It may be a dumb question, but is it not possible to use a different thread for the network generation, other than the main thread?
In such a way, even if the network generation takes long, it would not affect the main thread (and the poll block of existing components).
It would not be a definitive fix (the network generation would still need to be improved), but would mitigate the effects. 
Does it make sense?

doudou, 2018-01-03:
> Does it make sense?

Yes. That's actually why it is implemented this way already ;-) I would have called it a design flaw otherwise ;-) The bug is that the generation thread is __starving__ the main thread. Which is the issue that needs to be investigated.

Ruby and Python (and some other VMs) actually run only one thread at a time. They still time-share a single CPU, but there seem to be an issue that I don't completely understand so far that leads to having multi-second starvation of the main execution thread.

joaobrittoneto, 2018-01-15:
Alright,
After reading a little bit about threads in ruby, it starts to make sense to me. 
As there is only one thread in Ruby running, the other threads needs to time-share the CPU, and the thread's priority indicates which thread should run in the sequence. 
So the generation thread (`syskit-async-resolution`) probably has a higher priority than the main thread (`MAIN`), or has methods that take long.

I tried the `roby-log timepoints` in the same log.
The first field, I guess, is the time after start. The second field is not clear for me, but I guess the third field is the duration of the respective block, is that right?
So the 751 is equivalent to 16:28:51.
So the duration of a cycle (which call a pool block), from start to end, in normal case is ~0.1s, as expected.
`752.002 0.000 0.100   cycle:end`

But  then, there are two cycles that take longer.
`752.549 0.000 0.547   cycle:end`
`755.907 0.000 3.358   cycle:end` 

What seems to take really long in the second cycle:
`752.979 0.429 0.429       TCPServer#process_pending_requests:end`
`754.205 1.201 1.201       syskit:update_task_states:end`
`755.766 0.006 1.543         syskit:apply_requirement_modifications:end`

What does a cycle represent?
I have the feeling that it is not a priority issue, since the pool block is called every cycle.

```
751.502 0.000         cycle:start
751.502 0.000           process_events:start
751.502 0.000             workers
751.502 0.000             syskit-transformer transformer broadcaster start:start
751.503 0.000 0.000       syskit-transformer transformer broadcaster start:end
751.503 0.000             each_cycle:start
751.503 0.000 0.000       each_cycle:end
751.503 0.000             TCPServer#process_pending_requests:start
751.503 0.000 0.000       TCPServer#process_pending_requests:end
751.503 0.000             syskit:update_deployment_states:start
751.503 0.000 0.000       syskit:update_deployment_states:end
751.503 0.000             syskit:update_task_states:start
751.510 0.007 0.007       syskit:update_task_states:end
751.510 0.000             poll block of Dive::Compositions::Communication::IoDebug::Process:0x83bdca8<id:1912>(command_line: ["dumpcap", "-i", "any", "-q", "-f", "ip host 192.168.42.5", "-f", "port 49155", "-w", "debug_usbl.pcap"], working_directory: /home/flatfish/dev.bir/bundles/flat_fish/logs/20171205-1616):start
751.510 0.000 0.000       poll block of Dive::Compositions::Communication::IoDebug::Process:0x83bdca8<id:1912>(command_line: ["dumpcap", "-i", "any", "-q", "-f", "ip host 192.168.42.5", "-f", "port 49155", "-w", "debug_usbl.pcap"], working_directory: /home/flatfish/dev.bir/bundles/flat_fish/logs/20171205-1616):end
751.510 0.000             poll block of FlatFish::Tasks::Safety:0x5f3f660<id:2055>(auv_messaging: syskit_auv_messaging_def of profile:FlatFish::Actions::Live::Safety::Profile(), coms_network: network_communication_def of profile:FlatFish::Actions::Live::Safety::Profile(), coms_surface: surface_communication_def of profile:FlatFish::Actions::Live::Safety::Profile(), coms_uw: uw_communication_def of profile:FlatFish::Actions::Live::Safety::Profile(), flatfishd_com: FlatFish::Actions::Live::Safety.flatfishd_communication()):start
751.510 0.000 0.000       poll block of FlatFish::Tasks::Safety:0x5f3f660<id:2055>(auv_messaging: syskit_auv_messaging_def of profile:FlatFish::Actions::Live::Safety::Profile(), coms_network: network_communication_def of profile:FlatFish::Actions::Live::Safety::Profile(), coms_surface: surface_communication_def of profile:FlatFish::Actions::Live::Safety::Profile(), coms_uw: uw_communication_def of profile:FlatFish::Actions::Live::Safety::Profile(), flatfishd_com: FlatFish::Actions::Live::Safety.flatfishd_communication()):end
751.510 0.000             poll block of FlatFish::Tasks::FlatfishdCommunication:0xf307898<id:2173>(flatfishd_port: 49201, heartbeat_period: 1, housing: nav, local_port: 49200):start
751.511 0.000 0.000       poll block of FlatFish::Tasks::FlatfishdCommunication:0xf307898<id:2173>(flatfishd_port: 49201, heartbeat_period: 1, housing: nav, local_port: 49200):end
751.511 0.000             poll block for Dive::Compositions::Communication::ControlSourceLevel:0xee46d90<id:2508>(depth_threshold: -1, surface_threshold: -0.5):start
751.511 0.000 0.000       poll block for Dive::Compositions::Communication::ControlSourceLevel:0xee46d90<id:2508>(depth_threshold: -1, surface_threshold: -0.5):end
751.511 0.000             poll block for FlatFish::Compositions::SyskitAUVMessaging:0xedc1618<id:2510>(local_endpoint: AUV_MSG_ENDPOINT_ROCKENV_NAV, state_period_fast: 2, state_period_slow: 10):start
751.511 0.000 0.000       poll block for FlatFish::Compositions::SyskitAUVMessaging:0xedc1618<id:2510>(local_endpoint: AUV_MSG_ENDPOINT_ROCKENV_NAV, state_period_fast: 2, state_period_slow: 10):end
751.511 0.000             poll block of OroGen::Transformer::Task:0x80355b8<id:2488>(conf: ["default"], orocos_name: transformer_broadcaster):start
751.511 0.000 0.000       poll block of OroGen::Transformer::Task:0x80355b8<id:2488>(conf: ["default"], orocos_name: transformer_broadcaster):end
751.511 0.000             poll block for FlatFish::Tasks::RequireAutonomousMode:0x4f33910<id:3210>(period: 5, timeout: 10):start
751.511 0.000 0.000       poll block for FlatFish::Tasks::RequireAutonomousMode:0x4f33910<id:3210>(period: 5, timeout: 10):end
[... cut ...]
755.937 0.000 0.000         syskit:apply_requirement_modifications:end
755.937 0.000 0.003       event_propagation_phase:end
755.937 0.000             error_handling_phase:start
755.937 0.000               structure_check
755.937 0.000               exception_propagation
755.938 0.001 0.001       error_handling_phase:end
755.943 0.005             garbage_collect
755.943 0.000 0.036     process_events:end
756.000 0.057           sleep
756.000 0.000           at_cycle_end:start
756.002 0.001 0.001     at_cycle_end:end
756.002 0.000 0.095   cycle:end
```

doudou, 2018-01-15:
The durations do not take into account time spent in other threads. So, if the execution spends 2.5s in another thread between the cycle:start and cycle:end event, the duration will include these 2.5s in the main thread as well. That's what I believe happened (from the timepoint_* events in the first log).

One great way to visualize the timepoint log is to convert it to CTF and then read it with trace compass (http://tracecompass.org/)

The conversion is done with `roby-log timepoints -ctf /path/to/log`. This creates two things:
- a directory in /path/to/log.ctf
- a file named /path/to/log.ctf.names

Load the directory in trace compass by right-clicking on the Traces in the project explorer and then Right Click -> Open Trace.

If you look at the Call Stack view (at the bottom), you'll see that the names are dumb (0x1, ...). This is what the name file is for. The second button in the call stack view's toolbar will allow you to "select a mapping file" (select the .names file). The names will be resolved, and you'll be able to see how the different processes overlay each other.