rock-core/tools-syskit #163: Improve handling of backward-compatible behaviors

doudou, 2018-03-27:
For reference: https://github.com/rock-core/tools-syskit/issues/162

The problem here is how to smoothly update Syskit applications for things that would _in fine_ be backward-incompatible changes.

My proposal would be be to introduce fine-grained "optional backward-compatible behaviors", something like the CMake policies. We already have the orogen backward-compatible naming, which is hidden behind some "general" backward_compatible_naming flag but could be more explicitly documented this way. If some of the behaviors are enabled, Syskit could report them at startup (maybe with a 6 months deadline ?)

@dbcesar @2maz @alcantara09 @g-arjones any opinions ?

g-arjones, 2018-03-27:
Yeah, that's a tough one. In principle, I think the change is welcoming, leads to cleaner codes on the app side and probably eases maintenance on syskit's side.

> If some of the behaviors are enabled, Syskit could report them at startup (maybe with a 6 months deadline ?)

This sounds reasonable, the catch is that these warnings will probably get lost between the thousands of other warnings one gets when a syskit instance is ran. That said, I see no better way.

As far as BIR is concerned, things are still running on those frozen *-before-test-refactoring branches. Currently, I am not in a position to say if there are plans to update (@joaobrittoneto, @gustavoneves12 ?) but I guess it is unlikely.

doudou, 2018-03-27:
> This sounds reasonable, the catch is that these warnings will probably get lost between the thousands of other warnings one gets when a syskit instance is ran.

Yeah, for sure ... However, it i's actually possible to run a Syskit instance warning-free ;-)

Another behaviours that could be implemented: a warning window once a month when starting the syskit IDE + warning when the "expiration date" passes

Note that the warning could be shown whenever *any* Syskit command is executed...

g-arjones, 2018-03-28:
> Note that the warning could be shown whenever any Syskit command is executed...

:+1:

2maz, 2018-03-28:
> If some of the behaviors are enabled, Syskit could report them at startup (maybe with a 6 months deadline ?)

If I understand correctly what happens after the deadline is: behaviours remain enabled, but warnings disappear. I would probably leave out the deadline, but rather introduce a corresponding flag (or gcc-like set of flags) to enable/disable such warnings.

Anyhow ok for me.

doudou, 2018-03-28:
> If I understand correctly what happens after the deadline is: behaviours remain enabled, but warnings disappear

My idea would be
1. on introduction of the NEW behavior, the new behavior is turned off by default (and warnings start to appear). Existing bundles only see the warning appear. New bundles, as created by `syskit gen`, switch the the NEW behavior by default, by turning it ON explicitly in `config/init.rb`
2. on the deadline, the old behavior can be removed altogether, and trying to set the backward-compatible config flag to `false` will raise. Setting it to `true` is OK.

The `common_models` bundle should be compatible with both old and new behaviors at the same time. This is something that we might need to revisit as we go along, but that's the intended goal.

Note that because I'd like to stick with SemVer, the second deadline is an "earliest" and not exact. The old behavior will be removed on the next major release after this date.