rock-core/tools-syskit #187: `orogenconf` will fail if a component interface is now incompatible with an existing configuration file

doudou, 2018-06-22:
It seems that orogenconf actually loads the configuration files, which makes it fail if a component's
configuration file is invalid because the interface changed). This is obviously a critical use-case
(one will want to geneate a new config file).

It might apply only to the config file(s) for the orogenconf's argument. I only tried this one.

~~~
dev@heads:~/dev/bundles/heads/config/orogen/live$ syskit gen orogenconf -r live motors_elmo_ds402::ReaderTask
WARN: integrating typelib plugin using the TYPELIB_RUBY_PLUGIN_PATH environment variable is deprecated
WARN: just put a file called typelib_plugin.rb into a subfolder from the RUBYLIB (e.g. base/typelib_plugin.rb)
WARN: offending dir: /home/dev/dev/install/share/typelib/ruby
Bundles[INFO]: Active bundles: heads
/home/dev/dev/tools/orocos.rb/lib/orocos/configurations.rb:426:in `block in normalize_conf': error loading /home/dev/dev/bundles/heads/config/orogen/motors_elmo_ds402::ReaderTask.yml: while loading section default factors is not a property of motors_elmo_ds402::ReaderTask (Orocos::TaskConfigurations::ConversionFailed)
        from /home/dev/dev/tools/orocos.rb/lib/orocos/configurations.rb:422:in `each'
        from /home/dev/dev/tools/orocos.rb/lib/orocos/configurations.rb:422:in `normalize_conf'
        from /home/dev/dev/tools/orocos.rb/lib/orocos/configurations.rb:229:in `block in load_from_yaml'
        from /home/dev/dev/tools/orocos.rb/lib/orocos/configurations.rb:217:in `each'
        from /home/dev/dev/tools/orocos.rb/lib/orocos/configurations.rb:217:in `load_from_yaml'
        from /home/dev/dev/tools/syskit/lib/syskit/task_configuration_manager.rb:108:in `load_from_yaml'
        from /home/dev/dev/tools/syskit/lib/syskit/task_configuration_manager.rb:94:in `block in reload'
        from /home/dev/dev/tools/roby/lib/roby/app.rb:1524:in `isolate_load_errors'
        from /home/dev/dev/tools/syskit/lib/syskit/task_configuration_manager.rb:91:in `reload'
        from /home/dev/dev/tools/syskit/lib/syskit/models/task_context.rb:196:in `configuration_manager'
        from /home/dev/dev/tools/syskit/lib/syskit/roby_app/plugin.rb:263:in `block in project_define_from_orogen'
        from /home/dev/dev/tools/syskit/lib/syskit/roby_app/plugin.rb:255:in `each_value'
        from /home/dev/dev/tools/syskit/lib/syskit/roby_app/plugin.rb:255:in `each'
        from /home/dev/dev/tools/syskit/lib/syskit/roby_app/plugin.rb:255:in `map'
        from /home/dev/dev/tools/syskit/lib/syskit/roby_app/plugin.rb:255:in `project_define_from_orogen'
        from /home/dev/dev/tools/syskit/lib/syskit/roby_app/plugin.rb:186:in `block in default_loader'
        from /home/dev/dev/tools/orogen/lib/orogen/loaders/base.rb:451:in `block in register_project_model'
        from /home/dev/dev/tools/orogen/lib/orogen/loaders/base.rb:450:in `each'
        from /home/dev/dev/tools/orogen/lib/orogen/loaders/base.rb:450:in `register_project_model'
        from /home/dev/dev/tools/orocos.rb/lib/orocos/default_loader.rb:49:in `register_project_model'
        from /home/dev/dev/tools/orogen/lib/orogen/loaders/base.rb:441:in `register_project_model'
        from /home/dev/dev/tools/orogen/lib/orogen/loaders/base.rb:104:in `project_model_from_text'
        from /home/dev/dev/tools/orogen/lib/orogen/loaders/base.rb:88:in `project_model_from_name'
        from /home/dev/dev/tools/orogen/lib/orogen/loaders/aggregate.rb:63:in `block in project_model_from_name'
        from /home/dev/dev/tools/orogen/lib/orogen/loaders/aggregate.rb:58:in `each'
        from /home/dev/dev/tools/orogen/lib/orogen/loaders/aggregate.rb:58:in `project_model_from_name'
        from /home/dev/dev/tools/syskit/lib/syskit/roby_app/plugin.rb:480:in `using_task_library'
        from /home/dev/dev/tools/syskit/lib/syskit/cli/gen_main.rb:286:in `orogenconf'
        from /home/dev/.autoproj/gems/ruby/2.3.0/gems/thor-0.20.0/lib/thor/command.rb:27:in `run'
        from /home/dev/.autoproj/gems/ruby/2.3.0/gems/thor-0.20.0/lib/thor/invocation.rb:126:in `invoke_command'
        from /home/dev/.autoproj/gems/ruby/2.3.0/gems/thor-0.20.0/lib/thor.rb:387:in `dispatch'
        from /home/dev/.autoproj/gems/ruby/2.3.0/gems/thor-0.20.0/lib/thor/invocation.rb:115:in `invoke'
        from /home/dev/.autoproj/gems/ruby/2.3.0/gems/thor-0.20.0/lib/thor.rb:238:in `block in subcommand'
        from /home/dev/.autoproj/gems/ruby/2.3.0/gems/thor-0.20.0/lib/thor/command.rb:27:in `run'
        from /home/dev/.autoproj/gems/ruby/2.3.0/gems/thor-0.20.0/lib/thor/invocation.rb:126:in `invoke_command'
        from /home/dev/.autoproj/gems/ruby/2.3.0/gems/thor-0.20.0/lib/thor.rb:387:in `dispatch'
        from /home/dev/.autoproj/gems/ruby/2.3.0/gems/thor-0.20.0/lib/thor/base.rb:466:in `start'
        from /home/dev/dev/tools/syskit/bin/syskit:40:in `<main>'
~~~

1ADZX, 2020-06-01:
I have encountered similar problems. How did you finally solve them?

doudou, 2020-06-04:
By fixing the configuration file. Behavior should be improved during `orogenconf`, to simply ignore the invalid conf file (or, better, by loading none of the Syskit configuration, which are not needed during code generation)