rock-core/tools-syskit #252: No access to a dynamic input port to perform component test

geovanemimoso, 2020-05-28:
I have a component that the input ports are dynamic. 

The ports are created on the configure state according to component configs. While doing the unit tests for this component, I realized that even after the configuration of the ports are done, i could not find the proper interface to access the inputs. 

For instance, considering that I have: **in_x** and **in_y** as names for my dynamic inputs, after configuring the inputs,  i’m not able to find  the **task.in_x_port** and **task.in_y_port** (or the correspondent) to work with them like a "static" port would have.

I try to create the port on the test with : `task.orocos_task.create_input_port('in_x', '/type')`

But i got: 

`NoMethodError: undefined method create_input_port' for #<Orocos::TaskContext: Orocos::TaskContext//task_under_test>`

For now, the only way i could find to access the created input was by using: 

`task.orocos_task.in_x.write(samples_in)`

But I guess this is not a good practice and in some tests a small amount of the samples written never arrive what makes the aproch below not reliable. 

```
output = expect_execution { task.orocos_task.in_x.write(samples_in)}
              .to { have_one_new_sample task.out_port }
```

The way i managed to make it work properly right now, is by using the following approach: 

```
reader = task.orocos_task.out.reader
task.orocos_task.in_x.write samples_in
output = expect_execution.to { have_one_new_sample reader }
```
 
Do you have any guideline on how i should properly access a dynamic input port using the syskit orogen-test??

doudou, 2020-05-30:
You're touching the problem of accessing dynamic ports in Syskit in general, which is still undocumented :( and will actually would be painful to use in the syskit tests (a.k.a. we'd have to implement helpers for them to make them reasonable to use).

I don't have the time to take care of it right now, unfortunately, so in the meantime:

Don't use port.write. It's a mis-feature and is unreliable. You can use `syskit_write` with a "raw" port:

~~~ ruby
output = expect_execution { syskit_write task.orocos_task.in_x, samples_in }
              .to { have_one_new_sample task.out_port }
~~~

As for `create_input_port`, well, it couldn't work: the Ruby layer can't tell your C++ component to create ports this way.

geovanemimoso, 2020-06-01:
Thanks for the Information @doudou! :)

doudou, 2020-06-04:
Thanks for reporting. I'll leave the issue open, this is definitely something that should be both documented and improved.