rock-core/tools-syskit #267: `have_new_samples` test predicate does not work with a timeout

g-arjones, 2020-10-30:
No samples are received when a timeout is set in the expectation.

To reproduce change this: https://github.com/rock-core/tools-syskit/blob/master/test/test/test_execution_expectations.rb#L138-L143

to:

```ruby
                it "passes if the task emits the required number of samples "\
                   "and returns them" do
                    value = expect_execution.timeout(1.0) { syskit_write task.in_port, 10, 20 }
                            .to { have_new_samples task.in_port, 2 }
                    assert_equal [10, 20], value
                end
```

g-arjones, 2020-10-30:
It's actually failing with `have_one_new_sample` as well...

doudou, 2020-10-30:
One is implemented with the other. I don't have time to look into it right now unfortunately.

g-arjones, 2020-12-06:
@doudou Looks like you are very invested in syskit right now, could you please take a look at this one?

doudou, 2020-12-07:
Saw your message, told myself that I would put that on my todo list for tomorrow, turned off the computer and then my brain just **saw it**. In

~~~
value = expect_execution.timeout(1.0) { syskit_write task.in_port, 10, 20 }
            .to { have_new_samples task.in_port, 2 }
~~~

you are passing the block to `timeout`, not `expect_execution`. A.k.a. it never gets executed. The general syntax is

~~~
expect_execution { ACTIONS }
   .option1.option2
   .to { PREDICATES }
~~~

Note that I do think this syntax is very awkward. I want to create a more assert-y type of syntax that would remove the need to chain so many things. I'm open for suggestions in doing so ...

g-arjones, 2020-12-07:
> you are passing the block to timeout, not expect_execution

Of course I am. That's... embarrassing.

> would remove the need to chain so many things

One thing that could clean up the API a bit was allow passing (some) of these options as keyword arguments to `#expect_execution`. That would probably avoid silly mistakes like this one.

To be honest though, what I see people struggling the most with is understanding what `#to` returns, specially when there are multiple expectations.

g-arjones, 2020-12-07:
BTW, feel free to close this.

Thanks!

closed 2020-12-07 by doudou